condac/openSimIO #17: Error loading X Plane 11 config.txt

aerodiduch, 2019-07-25:
Hi! 

I'm sure that this is not working because I missed something in the config/upload to the Arduino.

I have an Arduino Uno board. This is my first project with it. I followed all the steps, downloaded the source code and without any modification uploaded to the board, putting the plugin in the sim root folder and then making the config.text file as stated in the wiki. Script was upload with no problem to the board and then X Plane was initialized. When going to plugins>openSimIO toggle debug & send config this error appears. Like said before, its the default config.txt in the plugin folder. 

[image]

`# Hash tag is considered a comment, can only be in beginning of sentence
# You must fill in all fields even if analog scales is not used by digital inputs etc...

# masterId.slaveId.Pin;pinExtra;iomode;reverse;center;pinMin;pinMax;xplaneDataRef;xplaneMin;xplaneMax;xplaneExtra;

# example string:
# 1.0.D5;0;DI_INPUT_PULLUP;0;0;0;1;sim/cockpit/switches/pitot_heat_on;0;1;0;
# 1.0.A0;0;AI_RAW;0;500;100;900;sim/joystick/yoke_pitch_ratio;-1.0;1.0;0;

#/192.168.0.1 49000/ # this tells it to use ethernet for the moment

1.0.D7;0;DO_BOOL;0;0;0;0;sim/cockpit/switches/pitot_heat_on;0;1;0;
#1.0.D2;0;DO_TM1637_DEC;0;1.0;0;0;sim/cockpit2/gauges/indicators/compass_heading_deg_mag;0;1;3;

# set electric pins for display
#1.0.D4;0;DO_LOW;0;1.0;0;0;sim/cockpit2/gauges/indicators/airspeed_kts_pilot;0;1;0;
#1.0.D5;0;DO_HIGH;0;1.0;0;0;sim/cockpit2/gauges/indicators/airspeed_kts_pilot;0;1;0;

# Servo test
#1.0.D8;0;AO_SERVO;0;1.0;0;0;sim/cockpit2/gauges/indicators/airspeed_kts_pilot;0;1;3;

# PWM test
#1.0.D11;0;AO_PWM;0;1.0;0;0;sim/cockpit2/gauges/indicators/airspeed_kts_pilot;0;1;3;
a
# Rotary test
#1.0.D22;0;DI_ROTARY_ENCODER_TYPE1;0;10.0;0;0;sim/cockpit/radios/com1_freq_hz;0;1;23; ## Extra pin not used it will always use pin+1 as extra

# Slave board test
#1.1.A1;0;AI_FILTER;1;500;12;1024;sim/joystick/yoke_pitch_ratio;-1.0;1.0;0;
#1.1.D13;0;DO_BOOL;0;0;0;0;sim/cockpit/electrical/landing_lights_on;0;1;0;

# 4x4 key matrix test
#1.1.D2;6;DI_4X4;0;0;0;0;sim/cockpit/switches/pitot_heat_on;0;1;0;
#1.1.D2;2;DI_4X4;0;0;0;0;sim/cockpit/electrical/landing_lights_on;0;1;1;
#1.1.D2;13;DI_4X4;0;0.1;0;0;sim/flightmodel/controls/elv_trim;-1.0;1.0;2;`

condac, 2019-08-06:
This is all because I have added an extra field in the config file. Before xplaneMin there should be an xplaneCenter. It is missing in alot of examples. 
There are also other issues regarding outputs. Expect a fix and update on all examples and documentations today or tomorrow. There will be a release v0.1.2 also soon with other fixes regarding outputs.

condac, 2019-08-06:
Also I'm sorry I did not see this issue earlier, but I think I have managed to setup email notifications on new issues now so I can see them in the future ;)

condac, 2019-08-06:
@aerodiduch I have updated all examples I can find. Please download latest v0.1.2 release and try again. Also the config files now include a way to tell it what COM port you should connect to and the line for network ip is also changed. Before it only tried to connect to COM! on windows and /dev/ttyUSB0 on linux and if your device was on another port it would not work. 
Let me know if it works or if you encounter any other issues that might need better documentation.

PR #32 (2020-12-29): Dev
Major new feature, multiple masters. 
Fixes #8 
Fixes #17 
Fixes #26 
Fixes #29
closed 2020-12-29 by condac