condac/openSimIO #31: Missing plugin

84Park, 2020-02-23:
Hi Condac, I am trying out your code downloaded as a zip file (2020-02-23) on a MacBook Pro, MacOS 10.15.3. I tried to follow your wiki instructions, but X-Plane 11.41 does not sense any plugin with the OpenSimIO folder saved in the plugins directory:  /Desktop/Planes/X-Plane 11/Resources/plugins/OpenSimIO.

I downloaded the X-Plane 11 sdk and placed that with the MakeFile.
I located the macOS sdk using "xcodebuild -sdk -version" which is 
Path: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk

I then tried to compile the MakeFile. rs232.h does not check for MacOS so it chokes on windows.h, but that error was cleared up with:
#if IBM > 0
	#include <windows.h>
#endif

Then many errors associated with "signal". 
openSimIO.c:51:7: error: redefinition of 'signal' as different kind of symbol

Any help on trying to get the plugin to work would be appreciated.

condac, 2020-02-24:
I have not tried compiling this for Mac. It is simply because I do not own any mac hardware and do not know anyone that have it. But I have tried to use makefiles that have prepared for the compatibility for this project to be able to work on mac if someone can help and find bugs and compile it. 

The error on the variable signal must be something mac specific that conflicts with a variable or function name used in a library that gets imported. I find no other use of the name in my code. But this is very simple to fix I can just rename this variable, and I have just now if you download the master branch again.

84Park, 2020-02-25:
Thank you. I will get the new code and try to compile it.