condac/openSimIO #35: arduino leonard

mikeyboy49, 2020-06-24:
Hi I have done all the config changes but get this when trying to upload to Arduino Leonard..

"Arduino: 1.8.12 (Windows 10), Board: "Arduino Leonardo"

C:\Users\owner\Downloads\openSimIO-0.1.2\iotest\iotest.ino: In function 'void parsePCSerial()':

iotest:384:19: error: invalid initialization of reference of type 'HardwareSerial&' from expression of type 'Serial_'

       waitForData(pcSerial, 3);

                   ^~~~~~~~

C:\Users\owner\Downloads\openSimIO-0.1.2\iotest\iotest.ino:366:6: note: in passing argument 1 of 'void waitForData(HardwareSerial&, int)'

 void waitForData(HardwareSerial& inSerial, int nr) {

      ^~~~~~~~~~~

iotest:386:19: error: invalid initialization of reference of type 'HardwareSerial&' from expression of type 'Serial_'

       waitForData(pcSerial, 3);

                   ^~~~~~~~

C:\Users\owner\Downloads\openSimIO-0.1.2\iotest\iotest.ino:366:6: note: in passing argument 1 of 'void waitForData(HardwareSerial&, int)'

 void waitForData(HardwareSerial& inSerial, int nr) {

      ^~~~~~~~~~~

iotest:388:19: error: invalid initialization of reference of type 'HardwareSerial&' from expression of type 'Serial_'

       waitForData(pcSerial, 3);

                   ^~~~~~~~

C:\Users\owner\Downloads\openSimIO-0.1.2\iotest\iotest.ino:366:6: note: in passing argument 1 of 'void waitForData(HardwareSerial&, int)'

 void waitForData(HardwareSerial& inSerial, int nr) {

      ^~~~~~~~~~~

iotest:403:23: error: invalid initialization of reference of type 'HardwareSerial&' from expression of type 'Serial_'

           doSomething(pcSerial);

                       ^~~~~~~~

C:\Users\owner\Downloads\openSimIO-0.1.2\iotest\iotest.ino:476:6: note: in passing argument 1 of 'void doSomething(HardwareSerial&)'

 void doSomething(HardwareSerial& inSerial) {

      ^~~~~~~~~~~

exit status 1
invalid initialization of reference of type 'HardwareSerial&' from expression of type 'Serial_'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
"

Im not a programmer and now confused... Help!!!!

condac, 2020-06-25:
Arduino Leonardo uses a different type of serial interface because it have a software serial. It is listed as "not recommended" in supported hardware because of this. https://github.com/condac/openSimIO/wiki/Compatible-Hardware 
If you have an Uno or Mega please use them instead. 
I will make fixes so it might work on Leonardo in the future but at the moment there is not time to do so.

mikeyboy49, 2020-06-25:
Thanks for the update, I have an Uno so will try that...

closed 2020-07-16 by condac