condac/openSimIO #38: Serial config not working!

ryucz, 2020-09-25:
While trying to send config, i get message "Error! converting config line 1;s;\\.\COM1"

Additional research comes to this:

pins.c @ line 53 
`if((line[0] == '#') || (line[0] == '\r') || (line[0] == '\n') )`

is missing
`line[0] == '/'`

philhynes01, 2020-11-16:
I had this same issue, I added the missing "\" to the start of the string \1;s;\.\COM1 and it worked

posixPilot, 2020-11-29:
I patched pins.c like this, which is what I think ryucz is referring to:

53c53
<   if((line[0] == '#') || (line[0] == '\r') || (line[0] == '\n') ) {
---
>   if((line[0] == '#') || (line[0] == '\r') || (line[0] == '\n') || (line[0] == '/') ) {

condac, 2020-12-29:
I have not been able to see why this will prevent the program from working correctly, in my test it just throws an error message when reading the config but still operates normal. This is just hiding the error message and not changing the outcome so if the program is not working this will probably not fix the issue. 
But there should be no error messages during normal operations just a bit confused to why it would fix an issue.

posixPilot, 2020-12-30:
Hi Condac,

> But there should be no error messages during normal operations just a bit confused to why it would fix an issue.

You are probably correct, but I think any user would assume the system couldn't work with this error.  At least that is what I assumed.  I believe the serial port config is silently read in another part of the code, but I did not put the time in to find out how that works.  Yes, this does just hide the error, but the error is apparently not relevant to this part of the code anyways.  It is possible that the only thing this fixed was my perception of broken.

  Thanks
  Mike

closed 2021-12-14 by condac