condac/openSimIO #39: Plugin X-Plane not compile

JesusGonfer, 2020-12-29:
Hi, I discovered this xplane plugin and I found it very interesting since it is apparently easy to scale your code to add new functionalities. But I am having a problem compiling the repository plugin from my own computer.

  -- Issue --
The plugin compiles correctly but I can't run xplane once I add it, it gets stuck in the plugin loading process. On the other hand, the plugin that you offer compiled works perfectly so it is ruled out that it is a problem with my Xplane installation

-- Devices --
SDK: 2.1.3
OS: W10 (with UBUNTU subsystem)
in my ubuntu subsystem I have installed:
  `sudo apt-get install gcc-mingw-w64-x86-64 binutils-mingw-w64-x86-64`

-- OTHER THINGS --
I compile the plugin from my ubuntu subsystem using:
`./makeall.sh`

[image]

Any help is welcome and thanks in advance, if you need more information or I have explained myself wrong, I will try to do better next time

condac, 2020-12-29:
I have the same warnings in my compile if I remember correctly. The code have a few debug prints during initiation (XPLMDebugString("serial done"); for example). Look in the xplane log when you start and try to see what the last output from the plugin is.  And if neccesary add more debug prints to nail down where it crash?

JesusGonfer, 2020-12-29:
Reviewing trying to solve the problem I reinitiated my ubunty subsystem and now it doesn't seem to compile the command ` make ` or ` ./makeall.sh ` I definitely broke my environment, as soon as I have it operational again I will try what you tell me. 
[image]

Thank you very much for the speed in responding

JesusGonfer, 2020-12-29:
I managed to compile again but I still have the same problem, I attach the xplane log, where you can see that it does not even load the plugin, it "gets stuck" before not even loading it, as if it were badly compiled

Log file plugins section:
**_==== CHANGING LANGUAGE TO: Español. ==== 
Fetching plugins for C:\X-Plane 11/Resources/plugins
Loaded: C:\X-Plane 11/Resources/plugins/PluginAdmin/64/win.xpl (xpsdk.examples.pluginadmin)._**

On the other hand, if I add the plugin that you compiled, it works and I see the boot traces in the log.
Thank you very much and sorry for the inconvenience

ninorazpotnik88, 2021-01-26:
Hy.

Im noob in programming and i have to make opensimio to work.

I copied all files to plugin folder but still cant see plugin in game.
Do i have to compile something? where can i get .xpl file?

Greetings Nino

JesusGonfer, 2021-01-26:
If you only need the files for the simulator, you can find them already compiled in the [Release tab](https://github.com/condac/openSimIO/releases)

But in case you need to recompile the plugin you should install the appropriate dependencies

ninorazpotnik88, 2021-01-26:
Thanks for answer. I can see plugin now.

Now i have new problem, in config.text i set com port on 3 and later on 1.  I dont get any data on board. my board is on port 3.
I uncomented servo row. 

greetings

JesusGonfer, 2021-01-26:
I have never used the serial port, I have always used it for ethernet. My only recommendation is that you look at the examples. Good luck and have a nice day

closed 2021-02-03 by JesusGonfer