condac/openSimIO #41: I2C Communication master / slave ?

osair777, 2021-07-25:
Is the communication between Master and Slave I2C (SDA/SCL) or another one (e.g. TX-RX; SCK-MISO-MOSI).
At the moment, I can’t test it.
Thanks for the information.

condac, 2021-07-26:
It uses the serial ports RX/TX. On MEGA it have extra serials for this but on UNOs and others it use the port shared with the usb interface.

osair777, 2021-07-27:
Ok thanks. 
And with 2 or more Master Devices, I write 2 or more COM Ports in the config File or can i use RX / TX from the first Master Board?

condac, 2021-07-27:
You can use more master devices that connect to a computer with one usb each or ethernet shields. Or you can chain slave devices to a master device.  

The first number in the config file is master id and second number is slave id. 

Multiple masters:
```
 /1;s;/dev/ttyUSB0;
 /2;s;/dev/ttyUSB1;
 1.0.D13;0;DO_BOOL;0;0;0;1;sim/cockpit/switches/pitot_heat_on;0;0;1;0;
 2.0.D13;0;DO_BOOL;0;0;0;1;sim/cockpit/switches/pitot_heat_on;0;0;1;0;
```
using slave:
```
/1;s;/dev/ttyUSB0;
1.0.D13;0;DO_BOOL;0;0;0;1;sim/cockpit/switches/pitot_heat_on;0;0;1;0;
1.1.D13;0;DO_BOOL;0;0;0;1;sim/cockpit/switches/pitot_heat_on;0;0;1;0;
```
But be aware that multiple devices are still not tested much and might be buggy, but this is how it is intended to work at least. The real implementations I have with this currently is multiple ethernet masters with only one way signals from x-plane to external software, not arduinos. So it was some versions ago when I tested serial slaves.

osair777, 2021-07-27:
ok, that's how I thought. Thank you
So does that mean that I can connect 3 more slaves to a Mega, because there are only 3 more RX TX (1 to 3) without a USB serial port? And where can I set the slave numbers in the IOTEST? If I let the script assign this automatically, I don't know which number the respective slave has.

condac, 2021-07-27:
You can have how many slaves you want, from the master you connect the master TX -> RX slave1  TX slave1 -> RX slaveN TX slaveN -> master RX. So they are connected in a chain, they will get numbered by the order they are in the chain automatically. In the arduino "iotest" program you need to comment and uncomment the lines so it is defined as SLAVE and not MASTER and program the slaves. You also need the define SERIAL_CHAIN for this to work in both master and slaves, All these are in Conifg.h

osair777, 2021-08-17:
I now have one MEGA as a master and three UNOs as a slave as a test. that works so far or the Dararefs are read and processed. The only problem is that there is no output on slaves 2 and 3 (e.g. LED). Slave 1 and master work without problems. Even if the slaves are swapped with one another. It is always from the 2nd slave. Where is the problem?

[image]
[image]
[image]

condac, 2021-08-17:
Everything in your configuration looks correct, it is probably something that is not working correctly when the first slave receives the message and not forwarding it to the next.

osair777, 2021-08-17:
I reinstalled the plugin and iotest and reprogrammed the master and slaves, but without success. When I read out the master, all 3 slaves are recognized as such.
[image]