condac/openSimIO #46: Moving average filter

nandybx7160, 2025-05-03:
**Is your feature request related to a problem? Please describe.**
Input AI_FILTER function averages the sum of 10 readings of the input pin

**Describe the solution you'd like**
Replace this function using moving average library <movingAvg.h> which is faster. Or add another input type AI_MOVING_AVG using this library for evaluation.

**Describe alternatives you've considered**
No other alternatives

**Additional context**
None

condac, 2025-05-04:
Looking at the source code for movingAvg it does the exact same thing as my function but without the deadband. So I find it very hard to see why it will be faster. Memory is also an issue that needs to be considered, the movingAvg library will take extra program memory because it can do more than we use. Also for something that is this simple and already implemented I do not find the added dependency worth the problem. 

The main reason for using an average with deadband is to reduce the amount of analog signals that get updated every cycle. This is meant to be used for controls that do not need high precision and is just meant to be kinder to the network traffic. 
If you need to filter the signals for noise I would recommend to put that code on the receiving end to unload the Arduino from unnecessary cpu cycles if you find that you want to make the Arduino run faster. 

The end result if you use a filter without deadband will still be a update every cycle if the analog value is noisy and in that case we have made the work on the Arduino more cpu intensive for no gain that the receiving computer could do instead. 

The goal is to get the analog and digital signals unmodified as fast and easy to the computer, no calibrations and transformations is meant to be done on Arduino side unless it is absolutely necessary for technical reasons. The Arduino will have lot of things to do anyway. 

Please explain in what usecase you need to filter signals and what it is you are trying to accomplish if filtering on the reviving end can not be done. 

ps.
Nice to see that someone uses this program apart from me. I use this in 4 flight simulators with 2 Arduinos per simulator. So I will maintain this code for a long time to come, but it have not needed much change in a long time and have been very solid for our use case.

nandybx7160, 2025-05-05:
It's not exactly the same. In pins.c, there is a 10x loop to get the average for every reading. In movingAvg, at first, 10 samples are summed (and averaged) while every sample is stored in an array. For subsequent iterations, the 1) latest value is added to the running sum and 2) the oldest value is subtracted from the running sum  - no 10x loop. This is the only comparison I made. Btw, the filtering we use is called SMA (Simple Moving Average). I also found another method EMA (Expontial Moving Average) which involves a non-integer formula. I don't know whether performing this computation is lighter than the current routine - but it's more responsive to big changes in input than SMA (slower). But for muscular action, I think it makes no difference.

As for the deadband operation, I have not taken a close look how it works. But if the purpose is to minimize the network traffic, then it is great. 

I have not yet fully used openSIMIO - just an exploration with other Arduino-XPlane interfaces. I was interested in filtering of inputs to tame noise in axis controls like pitch. With my rudder pedal, I used a capacitor to help filter out the noise before I found movingAvg. 

I'm also looking for an interface that can trigger Xplane commands. I have not seen any example. Is it capable? Or it's a TODO feature? Tks for the response.

condac, 2025-05-09:
I think that you might be looking at the wrong code. For your case the function AI_OVERSAMPLE will do what you want. It is the same as AI_FILTERED but without the deadband. 
```
int currentState = analogRead(pin-DIGITAL_PIN_COUNT);
analogFilter[pin-DIGITAL_PIN_COUNT][analogFilter2[pin-DIGITAL_PIN_COUNT]] = currentState;
```
This part reads the current value and put it in an array at the position defined by `analogFilter2[pin-DIGITAL_PIN_COUNT]`. And by doing so we overwrite the oldest value in the array. Later in the code we do a ++ on this value to move to the next position in that array.  

Do you mean that the difference is that we do not store the latest value in the array but the latest average value instead? because in that case it will be very easy to just change the code to store sum in the array and not currentState.

And Yes, I can do Commands! Using a digital DI_INPUT_PULLUP if you just enter a command instead of a dataref it will set command. And you have 2 different behavior based on the xplaneextra setting, To do CommandBegin on true and End on false (reversible with the reverse option). Or if xplaneextra is set to 1 it will do CommandOnce.

condac, 2025-05-09:
I was not sure if we talked about the same thing with SMA, I am not that familiar with these terms. So i asked the hallucinating crap-bot called chatGPT ;) And according to him my code is a correct implementation of SMA, but it said I had an array overflow bug, so i fixed that now ;)

Also chatGPT said that my code is more efficient and uses less cpu cycles.

> Final Thoughts:
> Performance: In terms of CPU cycles, your approach with the running sum is more efficient. You avoid having to loop through the array to sum all values every time — instead, you simply add and subtract values, which results in fewer cycles.
> 
> Flexibility: The library code’s approach with a loop provides flexibility if you want to sum a different number of points (with nPoints). However, this comes at a slight performance cost.
> 
> Rounding: The library code rounds the average using (sum + nPoints / 2) / nPoints. This is a slight optimization compared to truncating the result, and it gives a more accurate average.
> 
> When Would This Matter?
> For small systems (like Arduino), minimizing loops and unnecessary calculations can have a real impact. Your approach, using the running sum, is more efficient.
> 
> If the system is large enough (i.e., not constrained by clock cycles or memory), then either method would perform similarly for small buffer sizes like nPoints = 10.

nandybx7160, 2025-05-15:
If the algorithm deletes the oldest value (or oldest sum) and inserting the latest value to avoid iteration, then it's the quickest way. The deadzone to minimize traffic is well and good. So, we can keep the routines AS-IS. But doing Commands plus options CommandBegin and CommandEnd are not well documented. I have to test your library again. Thank you.